opulo-inc/docs #19: Introducing openPNP basics like vision & feeders

InfiniteNesLives, 2022-07-09:
This has been a quite common stumbling block of users including myself.  I first had trouble with vision recognizing the homing fiducial and couldn't sort out how to resolve it until Bobbe asked in Discord if I had tried modifying the pipeline.  I had no clue what the vision pipeline was or how to get to it.  I've seen this same exact thing happen to each user in discord as they complete the lumen calibration.

I completely agree that it shouldn't be a goal of the lumen docs to teach users details about openpnp, that's the job of the openpnp docs which are quite good.  But the situation that occurs with everyone is that once you've completed the lumen calibration procedure and it's time to migrate over to openpnp and it's docs, you simply don't even have a clue as to all the things you don't know.  So if the lumen docs simply introduced the users to some of these topics with links to openpnp docs to learn more would help quite a bit.

Here's a list of some of the common stumbling blocks I see that users will have better luck skipping over with a list & links to openpnp:

1- vision pipeline: https://github.com/openpnp/openpnp/wiki/CvPipeline
    they even have a nice quick little video which had I found early on would have saved me from going deep into the wrong rabbit hole.. ;)  https://www.youtube.com/watch?v=iUEP0bILAU0

2- fiducials: https://github.com/openpnp/openpnp/wiki/Fiducials
  Might be good to suggest they test out finding fiducials on their PCB if present.  Might even be good to include in the homing calibration procedure as something to investigate if having issues as this is when I first had problems that needed to be solved with modifying a vision pipeline.

3- feeder setup:  https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration:-Feeders
   CurlyTaleGames had a good recommendation to suggest users setup their first static feeder: https://github.com/openpnp/openpnp/wiki/ReferenceStripFeeder 

4- Bottom vision:  https://github.com/openpnp/openpnp/wiki/Bottom-Vision
  advanced explanation of stages: http://openpnp.github.io/openpnp/master/org/openpnp/vision/pipeline/stages/package-summary.html

That's all I can think of at the moment.  I think just pointing these things out similar to how the lumen docs point out where to go for your first openpnp job would be of great help. :)

theacodes, 2022-07-09:
These would fit nicely into the "Next Steps" page in our OpenPNP docs. This would be a fantastic first issue for someone, and if someone wants to take it on feel free to reach out to me for assistance.

Memnarch1113, 2022-07-10:
I am working on this in the FTP guide: #11 , #10, #9, #8. Hopefully that will be enough knowledge for folks to get working even if they don't have the FTP PCB and are just starting with their own PCB.

gabeseltzer, 2022-10-03:
Those changes went live a while ago. I'm going to close this ticket as we've made a lot of progress. I'm going to work on updating the docs again right now for the new version of OpenPnP that improves how vision works, too. https://github.com/opulo-inc/docs/issues/41

closed 2022-10-03 by gabeseltzer