opulo-inc/docs #20: Make sure that the USB C receptacle feet are well soldered

theacodes, 2022-07-10:
Many community members have had issues with the USB C receptacle tearing off of the motherboard. We should add a note for folks to check and reenforce the receptacle's feet while they're soldering all the through-hole parts.

Memnarch1113, 2022-07-10:
Note: we did find the root cause of the bad USB C soldering and have resolved it. This was an assembly error at the factory. I don't know the exact date that we fixed the issue, but we haven't sent any new motherboards with the problem in a while, hopefully! 

On our internal issue tracker this is [II-88919](https://www.notion.so/opulo/USB-Connector-tore-off-e81cdfc08bd54459ad65c09b78b873a8)

@theacodes do you want to keep this open just in case there's more of them, or should we close since we think we fixed this? @sphawes FYI

For full transparency: we are replacing any motherboard that has a damaged USB C connector. At the moment we ask for just a photo that the connector really is damaged, but we don't ask for proof that the connector was not soldered correctly when it arrived (because nobody would have taken photos of the board before they started using it!). In the future, we may ask people to confirm which order they were and check the batch of motherboards to make sure that their particular case falls within the time period that we were making potentially faulty boards, but for now we're pretty trusting that any issues in the field are our fault, not user error on this problem.

theacodes, 2022-07-10:
I think it's still worthwhile for us to add this to the docs. Folks will already have their iron out and working on the board so giving the USB connector a quick check doesn't really cost any time or effort.

I'm glad that it's fixed going forward, but we're still seeing folks run into this in the Discord. In my opinion, the docs should give the user every chance of success and navigate them around pitfalls.

Yandrik, 2022-08-03:
[image]

Just to check, is this the correct part that needs to be reinforced? If so, I'd add it as an additional, optional step to the Mobo Soldering Docs

commit f0299796 (2022-10-10): fix #20
Make sure that the USB C receptacle feet are well soldered #20
closed 2022-10-25 by gabeseltzer