opulo-inc/docs #43: Clarify Documentation-to-Version Association

rheingoldheavy, 2022-10-11:
Just received my Lumen v3 kit, and went to the docs.opulo.io, clicked on the "Docs" link and was immediately taken to the "Lumen PnP Kit Assembly" page, which is where I've gone each time I've loaded the docs site. I had also almost purchased a bunch of filament to begin printing the parts needed for assembly. 

However, having now opened the box and seen that most of the parts already seem to be printed, and looked around and saw the "Lumen PnP Semi Assembly" in the top menu, I realized I've been looking in the wrong place the entire time. 

I would recommend renaming "Kit Assembly" and "Semi Assembly" to "Version 2" and "Version 3" since I don't see "semi assembly" anywhere on my packaging. A landing page that allowed you to go to the relevant documentation based on your version, rather than being sent straight to "Kit Assembly" would be helpful as well.

gabeseltzer, 2022-10-15:
Makes sense, I'll work on it!

gabeseltzer, 2022-12-29:
We've updated the QR code that's on the paper printed insert on new Lumens to point to the semi-assembled page :).

PR #68 (2022-12-29): redirect home docs button to semi assembly
fixes #43
closed 2022-12-29 by gabeseltzer