opulo-inc/docs #54: Bottom Camera Position

DavePaulsen, 2022-11-11:
Initial paragraphs in this section call out default Z-Heights as 61mm for V2 and 31.5mm for V3.

In section 5, "Set the Camera's Z-Axis locations:", Step 1 says for V2 set the values to 31.5mm and step 2 says for V3 set the value to 61mm.

BobDarlington, 2022-12-17:
Specific page with this issue: https://docs.opulo.io/openpnp/calibration/bottom-camera-position/
Ask me how I know.

gabeseltzer, 2022-12-20:
ah sorry, fixed this with the other issue. Thanks for the report!

BobDarlington, 2022-12-20:
Dude, you are the man.  Hey in the issue with the placement of the
bolts/wing nuts, I actually ended up using the original location but moved
the block with the 4 pneumatic lines.   Flipping like I originally planned
would've been fine too.   It's just tight in there.   Thank you very much
for attacking the documentation.  It's critical for people like me.   I
made it all the way to the FTP board and picking a resistor when openpnp
threw an error about unsafe z heights.  Just an FYI.  Trying to work around
this so I can send in an update to the process if at all possible so I can
be helpful and not a pest.

Thanks,
Bob