opulo-inc/docs #85: Suggestions regarding build instructions, V3

dalegrover, 2023-02-01:
LumenPNP build V3.0.2:

Most V3 instructions were clear and photos useful.  A couple of more important points are noted with ***.

*** Assembling the frame, step 17 and further:  These hammerhead t-nuts may not be intuitive--typically, the bolt can't be screwed far into them, or they can't rotate to engage the extrusion.  As shipped, the t-nuts are screwed too far on to rotate for some of the parts, and so must be loosened.

Related, in step 18, it may make sense to tighten the "bottom" bolt first (whose t-nut can't be seen); it is then obvious whether this engaged properly in the extrusion.  Then later the top bolt can be tightened; in this case, the t-nut can be seen to rotate (or not if it was too tight).

In Wiring, step 7:  Suggest noting that M5 bolt is going to be a tight fit through the cable chain.  (This may be reassuring.)

In Wiring, step 10 (and beyond):  Recommend being clear on the orientation of connectors.  You can probably use wire color (e.g., "with the black wire on the left").  In other cases, there are orientation tabs etc that can be called out.

General note:  I suggest adding a brief note early on describing how to safely unplug these electrical connectors (vs yanking out by wires), in case the user has made an incorrect connection.  Also, how to release the tubing by pushing down on the blue ring.  Neither may be obvious to some users.

In Wiring, step 13:  Clarify this is the unlabeled cable, since RM is also a 6 position/4-wire connector.  (Might also suggest the language "position" vs. "hole".)

*** In Wiring, step 15:  Cable is actually labeled LM, not LH (which is a tube).

*** In Wiring, step 17:  Cable is actually labeled RM, not RH (which is a tube).

In Wiring, step 24:  In my LumenPnP, the pneumatic tubing was very tight to the board.  No reason not to have some more slack, which would reduce the chance of putting force on the stepper driver heatsinks as the board shield is removed and installed.

In Wiring, step 25:  Note that the cable "clamp" (3D printed cable clamp that fits into the extrusion) may need to be shifted.

*** In Wiring, step 30:  XM was far too tight--I had to cut and re-do the zip ties on the strain relief print to get enough slack to route XM properly.  (This is not a documentation problem, but wiring harness problem.)

In Wiring, step 34:  Need to call out hardware sizes.  (BTW, I was missing a M5-10 bolt, only had one left.)

Additionally, it may be handy to provide a printed sheet with the various hardware types noted--in particular bolt lengths.

gabeseltzer, 2023-02-06:
@sphawes @lucian151 can you comment on:
> *** In Wiring, step 15: Cable is actually labeled LM, not LH (which is a tube).
> *** In Wiring, step 17: Cable is actually labeled RM, not RH (which is a tube).

Is this correct? I will have to photoshop the images if so.

lucian151, 2023-02-23:
RM/LM mean left motor or right motor, and plug into the left or right head. I can see why people are confused by this, so if the docs can be improved to address this fact it'd be easier than us updating cables to something else.

securelyfitz, 2023-04-19:
Was about to open an issue but i see many of my notes already recorded here.
I also had wires labeled RM and LM in steps 15/17; had to remove and re-ziptie the clamp in 29/30 ; and was missing a M5-10 in step 34.

One addition: in step 31, it shows the LH and RH labels on the pneumatic fitting restraint with LH on top. My 3-d printed restraint had RH on top. The labels on the tube were far enough up the tubes that they were hidden under the control board. So basically, it's impossible to get anything looking like the picture :)

I'll create a separate issue for the issues with the openpnp instructions.

gabeseltzer, 2023-05-16:
> *** In Wiring, step 15: Cable is actually labeled LM, not LH (which is a tube).
> *** In Wiring, step 17: Cable is actually labeled RM, not RH (which is a tube).

Photoshopped, thanks for the heads up.

> One addition: in step 31, it shows the LH and RH labels on the pneumatic fitting restraint with LH on top. My 3-d printed restraint had RH on top. The labels on the tube were far enough up the tubes [...]

I think I already photoshopped the docs page so it should match real life, thanks.

PR #111 (2023-05-17): Fixed Motor wire naming
* added clarification for how to attach wires
* other clarifications

Fixes #85
closed 2023-05-17 by gabeseltzer