opulo-inc/docs #89: change photo of bottom camera in Docs

Mitchwich, 2023-02-19:
Was talking with Stephen and noticed this photo is of an older model. the newer one is much more recessed. Could we update this photo to match what the customers are getting in the box currently?

gabeseltzer, 2023-02-21:
@sphawes are you able to get me a photo for this?

gabeseltzer, 2023-05-16:
@sphawes ping again, can you get me a photo? Thanks!

sphawes, 2023-05-23:
@gabeseltzer Added to the docs google photos drive, plus attached here! it's just a single pic without a nozzle tip above, if you need more specific images to do the swap lmk, but a pass through the docs seems like this should be an acceptable replacement for the existing close ups of the bottom cam
[image]

gabeseltzer, 2023-06-19:
There's a lot of photos of the bottom camera so I just decided to put a disclaimer early on that new units will look slightly different. Someday it would be nice to do fresh photos but I think confusion is minimal :)

PR #118 (2023-06-19): Explanation of recessed camera
fixes #89
closed 2023-06-19 by gabeseltzer