opulo-inc/docs #92: Needing step added to Dynamic Board Mounts setup

Mitchwich, 2023-03-15:
Hey Gabe,
  Stephen asked me to see if the steps on [THIS PAGE](https://docs.opulo.io/openpnp/ftp/2-setting-up-the-board/#mounting-the-ftp-pcb) showed the customer that they should line up the edge of the FTP with the top ridge on the Dynamic Board Mounts with the tabs on them. I couldn't find any steps that show this. Would we be able to add a step to showcase it? (The top ridge I'm talking about is the one on the tab that is normally pressed to release the FTP from the mounts.) Hope this makes sense, but if not hit me up on Discord and I can snap a pic.
  
  Thank you, Gabe!

gabeseltzer, 2023-03-26:
[image]
You mean this ridge? Is that just to make sure the FTP is properly seated, or is it to make sure it is square to the staging plate? @Mitchwich

Mitchwich, 2023-03-29:
You got it! Mainly, if you line the FTP up with that line, it will not have any play or wiggle once it is seated. @gabeseltzer (P.S. Thank you for working on this)

PR #110 (2023-05-17): Mounting instructions updated
Fixes #92
closed 2023-05-17 by gabeseltzer