opulo-inc/docs #112: Immediate crash starting up newest OpenPNP with feeder support for Opulo Feeder

buildingautomator, 2023-05-16:
Your suggested macOS dmg contains a version of OnePNP that crashes immediately with the following note about a plist value missing for camera approval and the trace (see Thread 3). See below.
So now I can't use my Opulo system at all anymore. Could you please fix this asap ?
Thanks.

Translated Report (Full Report Below)
-------------------------------------

Process:               JavaApplicationStub [4501]
Path:                  /Applications/OpenPnP/OpenPnP.app/Contents/MacOS/JavaApplicationStub
Identifier:            com.install4j.9204-4030-0867-1434.23
Version:               2023-04-05_08-24-36.0aa4ae8 (2023-04-05_08-24-36.0aa4ae8)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-05-16 15:14:17.9142 -0700
OS Version:            macOS 13.3.1 (22E772610a)
Report Version:        12
Anonymous UUID:        9F47936A-F745-C42C-ECA1-357C73A24F4D

Time Awake Since Boot: 9800 seconds

System Integrity Protection: enabled

Crashed Thread:        2  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace TCC, Code 0 
This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an com.apple.security.device.camera key with a string value explaining to the user how the app uses this data.

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x19baebf14 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x19bafe240 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x19baf4b78 mach_msg_overwrite + 604
3   libsystem_kernel.dylib        	       0x19baec290 mach_msg + 24
4   CoreFoundation                	       0x19bc0a8b8 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x19bc09198 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x19bc0858c CFRunLoopRunSpecific + 612
7   HIToolbox                     	       0x1a543ddf4 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x1a543dc30 ReceiveNextEventCommon + 648
9   HIToolbox                     	       0x1a543d988 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x19ee27f58 _DPSNextEvent + 636
11  AppKit                        	       0x19ee270f4 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
12  libosxapp.dylib               	       0x104a74218 -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 144
13  AppKit                        	       0x19ee1b558 -[NSApplication run] + 464
14  libosxapp.dylib               	       0x104a73fd0 +[NSApplicationAWT runAWTLoopWithApp:] + 208
15  libawt_lwawt.dylib            	       0x104ccc504 +[AWTStarter starter:headless:] + 488
16  libosxapp.dylib               	       0x104a75c48 +[ThreadUtilities invokeBlockCopy:] + 28
17  Foundation                    	       0x19cb9dd54 __NSThreadPerformPerform + 264
18  CoreFoundation                	       0x19bc0a710 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
19  CoreFoundation                	       0x19bc0a6a4 __CFRunLoopDoSource0 + 176
20  CoreFoundation                	       0x19bc0a414 __CFRunLoopDoSources0 + 244
21  CoreFoundation                	       0x19bc0901c __CFRunLoopRun + 828
22  CoreFoundation                	       0x19bc0858c CFRunLoopRunSpecific + 612
23  libjli.dylib                  	       0x102cd70dc CreateExecutionEnvironment + 400
24  libjli.dylib                  	       0x102cd34b0 JLI_Launch + 1212
25  JavaApplicationStub           	       0x100df8da8 -[Launcher launch] + 1880
26  JavaApplicationStub           	       0x100df933c launcher_main + 500
27  JavaApplicationStub           	       0x100df9468 main + 36
28  dyld                          	       0x19b7d3f28 start + 2236

Thread 1:
0   libsystem_pthread.dylib       	       0x19bb26d8c start_wqthread + 0

Thread 2 Crashed::  Dispatch queue: com.apple.root.default-qos
0   libsystem_kernel.dylib        	       0x19baf8d4c __abort_with_payload + 8
1   libsystem_kernel.dylib        	       0x19bb1de50 abort_with_payload_wrapper_internal + 104
2   libsystem_kernel.dylib        	       0x19bb1de84 abort_with_payload + 16
3   TCC                           	       0x1a07f27f0 __TCC_CRASHING_DUE_TO_PRIVACY_VIOLATION__ + 172
4   TCC                           	       0x1a07f3108 __TCCAccessRequest_block_invoke.203 + 588
5   TCC                           	       0x1a07f05ac __tccd_send_message_block_invoke + 632
6   libxpc.dylib                  	       0x19b871650 _xpc_connection_reply_callout + 124
7   libxpc.dylib                  	       0x19b871540 _xpc_connection_call_reply_async + 88
8   libdispatch.dylib             	       0x19b97c480 _dispatch_client_callout3 + 20
9   libdispatch.dylib             	       0x19b99a624 _dispatch_mach_msg_async_reply_invoke + 344
10  libdispatch.dylib             	       0x19b98ec04 _dispatch_kevent_worker_thread + 1280
11  libsystem_pthread.dylib       	       0x19bb280ac _pthread_wqthread + 344
12  libsystem_pthread.dylib       	       0x19bb26d94 start_wqthread + 8

sphawes, 2023-05-18:
Hi @buildingautomator! Thanks for the note, and I'm sorry for the issue. This is due to [an OpenPnP bug](https://github.com/openpnp/openpnp/issues/1559) about signing develop builds. Until the bug is fixed, there's a workaround that [I just added](https://github.com/opulo-inc/docs/commit/ccc48d638939b83957c5e2f292470d9919db8353) to the docs page that should be live in a few minutes [here](https://docs.opulo.io/feeders/3-software-update/software-update/). 

Please give it a try and let me know if it solves the problem for you!

gabeseltzer, 2023-06-03:
> So I went through the lengthy and difficult process to register with ST, download their firmware programmer only to find out that the Mac version doesn’t come with the USB drivers ????

I wasn't aware of this. I chose to mark the STM app as the default way to install new firmware on the LumenPnP because it's the easiest way to update the feeders, and I figured having one app to install both would be simplest. Unfortunately, I only tested on a PC, so I don't know of any additional complications on a mac. @sphawes do you happen to know of an easy way to get the drivers you need on a mac? 

@buildingautomator I can confirm it is possible to flash the firmware with a Mac or Linux computer, but I'm not 100% sure the best procedure. I'll look into it and get back to you, sorry for the delay!

buildingautomator, 2023-06-05:
No problem guys, just would have saved me a lot of time to have known.

What I did is, I installed the PC version on a Parallels virtual machine on an older Mac that was still running on Intel. I was then able to update.

Thanks for getting back to me,

Tony de Rijk
------------------
President & CEO
iDo Technology, Inc.

114 Larkfield Dr.
Santa Rosa, CA 95403, USA
Email ***@***.*** ***@***.***>

gabeseltzer, 2023-06-19:
https://github.com/opulo-inc/docs/commit/ccc48d638939b83957c5e2f292470d9919db8353 fixes this

closed 2023-06-19 by gabeseltzer