opulo-inc/docs #142: Vacuum sensor issues

bfj83, 2023-10-04:
M3426 G2 C1 I1 A110 always returns 0, if I change the address, it don't return anything, if i try M3426 G4 C1 I1 A110, it still returns 0

M3426 G2 C2 I1 A110 returns 26696, that don't change weather I have vacuum or not, same with sensor 1 that always reads 0

atanisoft, 2023-10-04:
what revision of the MB are you using?

bfj83, 2023-10-06:
it's MOBO REV 03

bfj83, 2023-10-06:
It doesn't have the interposer boards under the vacuum sensors

gabeseltzer, 2023-12-09:
@bfj83 I'm going to suggest you get in touch with our support directly [here](https://www.opulo.io/pages/contact-support). They can help more efficiently. Sorry to send you off in another direction but you'll get better help there.

closed 2023-12-09 by gabeseltzer