opulo-inc/docs #153: Mini version example

Utopiah, 2024-02-07:
Hi, amazing project and podcast!

I’d love to tinker with a LumenPnP but I’m just starting with electronics at home.

atanisoft, 2024-02-07:
Scaling the size down is relatively simple, change the length of the 2020 rails to fit the size you need and add a custom build plate surface. For the staging plate there is an OpenSCAD file you can use to generate the customized size you want [here](https://opulo.notion.site/LumenPnP-Community-Mods-ae0b8813479a42b79a91cda13b1b8c06#0764d89a30244256a0cee0ea1fe8439d).

Utopiah, 2024-02-07:
Super, thanks a lot for the clarification. What do you estimate would the smallest possible volume done this way?

atanisoft, 2024-02-07:
That would likely depend primarily on the number of components you have on your boards and how often you would want to reload strip feeders. But I wouldn't recommend smaller than ~300x300 (1/4 size of default) as you will have at least that amount of PnP components to fit on the bottom side of the staging plate(s) and will want space for strip feeders and space for your PCB to be assembled on the top side.

Utopiah, 2024-02-07:
> ~300x300

Do you mean 300mm? That'd be super compact, I'm surprise everything would still fit but that'd be perfect to start.

Also, very rough estimation, how much would the parts needed for such a mini version, say 1/4 as you suggest, costs? I imagine the minimum would be :

- smaller staging plate
- rails

... but I think that's it?

atanisoft, 2024-02-07:
> Do you mean 300mm?

Yes, it likely would be tight on placement of some of the components on the bottom side of the staging plate and require some creativity.

As for costs, that is very subjective based on where you source from and the quality of the parts you are sourcing.

closed 2024-09-11 by lucian151