opulo-inc/docs #161: "OpenPNP - Debugging" section has broken images/links

plaes, 2024-11-07:
This guide is broken: https://docs.opulo.io/openpnp/debugging

atanisoft, 2024-11-07:
this is a duplicate of https://github.com/opulo-inc/docs/issues/159 but you have provided more details on it than I delved into.

sphawes, 2024-11-12:
Thanks for the report @plaes and @atanisoft! I think this is the result of our new docs hosting, DigitalOcean, not redirecting urls without a trailing slash to one with a trailing slash. Github Pages does this automatically.

@theacodes lmk if you want access to our digitalocean account if that's helpful in figuring it out!

theacodes, 2024-11-13:
Okay, this was caused by two issues:

1. There were several instances of us improperly linking to other pages within MkDocs. MkDocs _really_ wants internal URLs to be in the form of `../relative/path/to/thing/index.md`. In this case, we'd linked to `../../debugging` instead of `../../debugging/index.md`. I have fixed _all_ occurrences of this in the codebase, which makes MkDocs create the correct link with trailing slash.
2. DigitalOcean does not automatically insert trailing slashes like GitHub Pages does ([ref1](https://github.com/slorber/trailing-slash-guide), [ref2](https://ideas.digitalocean.com/app-platform/p/trailing-slash-on-directories). GitHub Pages was actually saving us from our own mistakes, but DigitalOcean doesn't!

With internal links fixed, this should no longer be an issue when just navigating the documentation. Any links in the wild that are missing the trailing slash may still cause issues.

Fixed in e9191c4f4

closed 2024-11-13 by theacodes