opulo-inc/docs #168: I2C problem with vacuum sensor mcp3426

jfc54, 2025-05-23:
my machine is a lumenpnp 3.02 with marlin
FIRMWARE_NAME:Marlin bugfix-2.1.x(Oct. 3 2022 09:16:35)

I cannot get the vacuum sensor mcp3426 to work. I have tried all I2C addresses both through
openpnp and a python program (file with results attached).

Any help you can provide would be appreciated.
``import time
import datetime
import re

# Don't forget to pip install pyserial

import serial

# configure the serial connections (the parameters differs on the device you are connecting to)
ser = serial.Serial(
    #Change this eg. /dev/ttyUSB1
    port='/dev/ttyACM0',
    baudrate=115200,
    parity=serial.PARITY_NONE,
    stopbits=serial.STOPBITS_ONE,
    bytesize=serial.EIGHTBITS
)

ser.isOpen()

def string_to_bytes(str):
    b =  str.encode("utf-8")
    #print (f"b= {b}")
    return b

def run_gcode(cmd_str):
    #print("Sending gcode",cmd)
    #convert input string to bytes
    cmd = string_to_bytes(cmd_str)
    ser.write(cmd)
    
    #First reply is the result
    output=ser.readline()
    #print(output)
    #Now wait for an OK
    ok=''
    while(output!= b'ok\n' and ok != b'ok\n'):        
        ok=ser.readline()
    return output

# Run a GCODE to ensure Mobo is connected/awake
result = run_gcode('M114\n')
print (f"result= {result}")

for a in range(0,8):
    for c in range(1,3):
        for g in [1,2,4,8]:
            cmd = f"M3426 I1 C{c} G{g} A{a}\n"
            result = run_gcode(cmd)
            print (f"C{c} G{g} A{a} {result}")

'''
results:
result= b'X:0.00 Y:0.00 Z:0.00 A:720.00 B:720.00 C:10000.00 Count X:0 Y:0 Z:0 A:3197 B:3197 C:400000\n'
C1 G1 A0 b'Error:MCP342X i2c error\n'
C1 G2 A0 b'Error:MCP342X i2c error\n'
C1 G4 A0 b'Error:MCP342X i2c error\n'
C1 G8 A0 b'Error:MCP342X i2c error\n'
C2 G1 A0 b'Error:MCP342X i2c error\n'
C2 G2 A0 b'Error:MCP342X i2c error\n'
C2 G4 A0 b'Error:MCP342X i2c error\n'
C2 G8 A0 b'Error:MCP342X i2c error\n'
C1 G1 A1 b'Error:MCP342X i2c error\n'
C1 G2 A1 b'Error:MCP342X i2c error\n'
C1 G4 A1 b'Error:MCP342X i2c error\n'
C1 G8 A1 b'Error:MCP342X i2c error\n'
C2 G1 A1 b'Error:MCP342X i2c error\n'
C2 G2 A1 b'Error:MCP342X i2c error\n'
C2 G4 A1 b'Error:MCP342X i2c error\n'
C2 G8 A1 b'Error:MCP342X i2c error\n'
C1 G1 A2 b'Error:MCP342X i2c error\n'
C1 G2 A2 b'Error:MCP342X i2c error\n'
C1 G4 A2 b'Error:MCP342X i2c error\n'
C1 G8 A2 b'Error:MCP342X i2c error\n'
C2 G1 A2 b'Error:MCP342X i2c error\n'
C2 G2 A2 b'Error:MCP342X i2c error\n'
C2 G4 A2 b'Error:MCP342X i2c error\n'
C2 G8 A2 b'Error:MCP342X i2c error\n'
C1 G1 A3 b'Error:MCP342X i2c error\n'
C1 G2 A3 b'Error:MCP342X i2c error\n'
C1 G4 A3 b'Error:MCP342X i2c error\n'
C1 G8 A3 b'Error:MCP342X i2c error\n'
C2 G1 A3 b'Error:MCP342X i2c error\n'
C2 G2 A3 b'Error:MCP342X i2c error\n'
C2 G4 A3 b'Error:MCP342X i2c error\n'
C2 G8 A3 b'Error:MCP342X i2c error\n'
C1 G1 A4 b'Error:MCP342X i2c error\n'
C1 G2 A4 b'Error:MCP342X i2c error\n'
C1 G4 A4 b'Error:MCP342X i2c error\n'
C1 G8 A4 b'Error:MCP342X i2c error\n'
C2 G1 A4 b'Error:MCP342X i2c error\n'
C2 G2 A4 b'Error:MCP342X i2c error\n'
C2 G4 A4 b'Error:MCP342X i2c error\n'
C2 G8 A4 b'Error:MCP342X i2c error\n'
C1 G1 A5 b'Error:MCP342X i2c error\n'
C1 G2 A5 b'Error:MCP342X i2c error\n'
C1 G4 A5 b'Error:MCP342X i2c error\n'
C1 G8 A5 b'Error:MCP342X i2c error\n'
C2 G1 A5 b'Error:MCP342X i2c error\n'
C2 G2 A5 b'Error:MCP342X i2c error\n'
C2 G4 A5 b'Error:MCP342X i2c error\n'
C2 G8 A5 b'Error:MCP342X i2c error\n'
C1 G1 A6 b'Error:MCP342X i2c error\n'
C1 G2 A6 b'Error:MCP342X i2c error\n'
C1 G4 A6 b'Error:MCP342X i2c error\n'
C1 G8 A6 b'Error:MCP342X i2c error\n'
C2 G1 A6 b'Error:MCP342X i2c error\n'
C2 G2 A6 b'Error:MCP342X i2c error\n'
C2 G4 A6 b'Error:MCP342X i2c error\n'
C2 G8 A6 b'Error:MCP342X i2c error\n'
C1 G1 A7 b'Error:MCP342X i2c error\n'
C1 G2 A7 b'Error:MCP342X i2c error\n'
C1 G4 A7 b'Error:MCP342X i2c error\n'
C1 G8 A7 b'Error:MCP342X i2c error\n'
C2 G1 A7 b'Error:MCP342X i2c error\n'
C2 G2 A7 b'Error:MCP342X i2c error\n'
C2 G4 A7 b'Error:MCP342X i2c error\n'
C2 G8 A7 b'Error:MCP342X i2c error\n'

'''

theacodes, 2025-05-24:
Howdy, what motherboard are you working with? Can you share a picture of it?

jfc54, 2025-05-24:
Lumenpnp mobo rev 04 e224f64
picture to follow later today

jfc54, 2025-05-24:
[image]

theacodes, 2025-05-29:
Thank you for that. I think only the earliest motherboards used analog pressure sensors- REV4 definitely uses [digital ones](https://cfsensor.com/product/static-pressure-sensor-xgzp6857d/). To talk to them, you'll need to use the [I2C mux](https://www.nxp.com/docs/en/data-sheet/PCA9545A_45B_45C.pdf) since both sensors have the same I2C address.

An example of reading the sensor would look something like this:

```gcode
; Setup the I2C mux.

; It's at address 112 (0x70)
M260 A112
; Sending 0x01 enables channel 1, which lets us talk to VAC1  
M260 A112 B1
M260 S1 ; Send

; Ask the sensor to take a measurement.

; It's at address 109 (0x6D)
M260 A109
; First byte is the register address - 48 is the command register
M260 A109 B48
; Second bytes is the register value - 10 asks for a combined (pressure + temperature) reading
M260 A109 B10
M260 S1 ; Send

; Read back measurement from sensor. It's a 24-bit value spread across 3 registers (6, 7, 8).

; Read first byte (register 6)
M260 A109
M260 A109 B6
M260 S1 ; send read command
M261 A109 B1 S1 ; read one byte response
; echo:i2c-reply: from:109 bytes:1 data:00

; Read second byte (register 7)
M260 A109
M260 A109 B7
M260 S1
M261 A109 B1 S1
; echo:i2c-reply: from:109 bytes:1 data:04

; Read third byte (register 8)
M260 A109
M260 A109 B8
M260 S1
M261 A109 B1 S1
; echo:i2c-reply: from:109 bytes:1 data:d2
```

You'll need to take the 3 bytes use [the datasheet](https://cfsensor.com/wp-content/uploads/2025/02/XGZP6857D-Pressure-Sensor-V2.7.pdf) to decode them into a pressure value. Below is an example Python driver for this sensor that might prove useful- it's from a currently internal codebase, but hopefully is clear enough to get you going.

```python
# Copyright (c) 2024 Opulo, Inc
# Published under the Mozilla Public License
# Full text available at: https://www.mozilla.org/en-US/MPL/

"""Driver for communicating with the XGZP6857D i2c vacuum sensor.

See https://cfsensor.com/wp-content/uploads/2022/11/XGZP6857D-Pressure-Sensor-V2.5.pdf
"""

import enum
import time

from .i2cio import II2CIO, I2CError

_I2C_ADDR = 0x6D
_REG_DATA = (0x06, 0x07, 0x08)
_REG_TEMP = (0x09, 0x0A)
_REG_CMD = 0x30
_REG_SYS_CONFIG = 0xA5
_REG_P_CONFIG = 0xA6

_CMD_START_CONVERSION = 0b1000
_CMD_COMBINED_CONVERSION = 0b0010
_CMD_SLEEP_CONVERSION = 0b0011

class PressureRange(enum.IntEnum):
    K_RAW = 1
    K_500_TO_5000_KPA = 8
    K_260_TO_500_KPA = 16
    K_130_TO_260_KPA = 32
    K_65_TO_130_KPA = 64
    K_32_TO_65_KPA = 128
    K_16_TO_32_KPA = 256
    K_8_TO_16_KPA = 512
    K_4_TO_8_KPA = 1024
    K_2_TO_4_KPA = 2048
    K_1_TO_2_KPA = 4096

class XGZP6857D:
    def __init__(self, i2c: II2CIO, pressure_range: PressureRange = PressureRange.K_RAW):
        super().__init__()
        self.i2c = i2c
        self.pressure_range = pressure_range

    def read_pressure(self) -> float:
        """Read the current pressure from the device. The result is in Pascals."""
        io = self.i2c

        # Configure measurement & start conversion
        io.send(
            addr=_I2C_ADDR,
            data=bytes([_REG_CMD, _CMD_START_CONVERSION | _CMD_COMBINED_CONVERSION]),
        )

        # Wait 20ms for the measurement according to datasheet. Alternatively, we
        # could readback the 0x30 register and check for bit 3 to be clear, but by
        # the time Marlin does all that it'll probably have been 20ms anyway, so, 🤷‍♀️
        time.sleep(0.02)

        # The result is a 24-bit value measurement stored across three registers,
        # and the chip's i2c protocol makes us read them one at a time.
        result_bytes = []

        for reg in _REG_DATA:
            resp = io.transaction(
                addr=_I2C_ADDR,
                request_data=bytes([reg]),
                response_size=1,
            )

            if not resp:
                msg = "Could not read byte from i2c response"
                raise I2CError(msg)

            result_bytes.append(resp[0])

        # Reconstruct the 24-bit value, if the highest bit is set, then it's
        # negative.
        result: int = result_bytes[0] << 16 | result_bytes[1] << 8 | result_bytes[2]

        if result & (1 << 23):
            result = result - 2**24

        # Use the pressure range factor to convert to Pa
        return result / self.pressure_range

```

I'm going to close this issue for now, but let me know if you have any further questions.

jfc54, 2025-05-30:
Great. Just what I needed. Thank you.